jcmvbkbc/linux-xtensa #1: Xtensa HiFi 2/4

hackikg, 2021-06-15:
Hello Max, 
Please help me - full understood support procedure  add new audio codec decoder for Xtensa HiFi DSP. 
If I.MX8 SOC support Tellisica HIFi4 core, it's possible add support new codec Dolby AC-4.  

S 
Приветствую Макс, 
Было отлично поговорить по русски. 
Пытаюсь найти решение, чтобы SOC была возможность добавить поддержку Dolby AC-4 для использования одном из проектов по ATSC 3.0. Tellisica HIFi заявлена поддержка https://ip.cadence.com/ipportfolio/tensilica-ip/audio?cmp=pr091216_dolbyac4

jcmvbkbc, 2021-08-06:
@hackikg This repository is for xtensa linux, i.e. linux running on xtensa cores.
Using xtensa coprocessors found in SoCs to run codecs is not in scope of this project.

You may take a look at what Sound Open Firmware project does.

closed 2021-08-06 by jcmvbkbc

guglovich, 2022-07-06:
I wrote to SOF not too long ago about Xtensa HiFi 2 support. They replied they can do it, but only if you leave them a commit. But I'm not a developer.