jcmvbkbc/linux-xtensa #3: ESP32 UART RX support

max1220, 2022-10-06:
Currently, the UART driver only support transmitting(TX).
Could you implement a proper driver that can also handle receiving(RX) on the UART?

I've [implemented](https://github.com/max1220/linux-xtensa/commit/fd8080c29ae51d4fc90d07d680db2ea797ef68a6) a *very basic* timer-based driver that works well enough to type a few commands(But is probably bad in various ways; I've never modified a kernel driver before). I've also added a IRQ driver, but never managed to get that to work(crashes on boot).

jcmvbkbc, 2022-10-13:
Hi @max1220. Getting an issue in this repository (as opposed to a mailing list) was a bit unexpected, so I missed it.
Yes, I'll take a look at adding RX side to the esp32 UART driver, adding driver for the esp32 IRQ controller and using IRQ instead of polling where possible. Sometime this month, hopefully.

max1220, 2022-10-14:
I'll post further patches or questions, if any, to the mailing list in the future.

closed 2022-10-14 by max1220

jcmvbkbc, 2022-10-24:
@max1220 I've pushed the branch xtensa-6.1-esp32 with UART RX support and IRQ controller.