jcmvbkbc/linux-xtensa #4: How to build for ESP32?

zap8600, 2022-12-02:
Hi. I was wondering how to build Linux for the ESP32 and how to flash and boot it on an ESP32. How do I do so?

raspiduino, 2022-12-03:
I think it's almost impossible to do this, since esp32 have less RAM than a minimum Linux kernel requires

zap8600, 2022-12-03:
I'm talking about the Xtensa ESP32. There is a branch for it.

raspiduino, 2022-12-03:
Wow I have never known that. I just look at it and found `esp32_defconfig`

It would be awesome to see Linux run natively on esp32 xtensa. Previous Linux on esp32 are just Linux on emulator on esp32

zap8600, 2022-12-03:
I found a comment on a hackaday post. If it is true, then it is possible. [image]

raspiduino, 2022-12-03:
Holy cool

zap8600, 2022-12-03:
I'll see if a couple of hours more searching can tell me how to build it and how to flash and boot it.

raspiduino, 2022-12-03:
Have you tried building like you would do with normal Linux? Like getting a toolchain, setting `CROSS_COMPILE` and `ARCH` variables, then make defconfig?

jcmvbkbc, 2022-12-03:
> i'm not sure about flashing

See http://wiki.osll.ru/doku.php/etc:users:jcmvbkbc:linux-xtensa:esp32 for the flashing tips.
Also the kernel needs a compiler with -mforce-l32 support.

raspiduino, 2022-12-03:
Wow
Gonna try that in qemu-esp32 (because I have no real esp32:))

max1220, 2022-12-03:
A shameless bit of self-advertising, but I have some scripts that might help some of you:
https://github.com/max1220/xtensa-fun

I use these scripts to build and test Linux for the ESP32 in QEMU(I don't have any real hardware with PSRAM).
YMMV, it's more a little hack, but it has all the stuff you need(build toolchain, kernel, buildroot, bootloader, application, generate flash image).

I can't wait for proper FDPIC support(FDPIC are "real" ELFs and also should save some RAM, but it's a toolchain thing), because I think bFLT(Flat binaries, no/limited dynamic linking) is somewhat [broken](https://github.com/jcmvbkbc/linux-xtensa/commit/0eb21ca4a4f22160b25ccdfd884260430b036ee2)(or I'm too stupid), but I've got an image that boots to a shell just fine!

raspiduino, 2022-12-03:
Thanks for your useful repo!

Btw I saw llvm with xtensa support, maybe it can generate FDPIC xtensa?

max1220, 2022-12-03:
> Although I only have esp8266 and I don't think it will boot Linux **natively** (we are not talking about running an emulator on it) due to lack of RAM, but please tell me how to boot Linux natively on esp8266 if you know. Thank you!

You're right, the ESP8266 will likely* never natively boot a normal noMMU Linux: AFAIK It lacks support for mapping external memory, and doesn't have enough internal memory to boot Linux.

(You could theoretically *maybe* implement some sort of swapping, probably not using the regular swapping mechanism, and use an SPRAM or other external memory, even an SD-card. You'd need to swap out kernel memory segments as well, which is not something Linux is built for. But technically it should be possible!).

At that point performance would also probably be very bad, and you might as well consider an emulator, which can be
way more clever about how it uses memory, and could easily swap out to an SPRAM or SD-card. But [that's](http://dmitry.gr/index.php?r=05.Projects&proj=07.%20Linux%20on%208bit) [been](https://github.com/cnlohr/mini-rv32ima) [done](https://github.com/juiceRv/JuiceVm) before. 

* (It might be possible that in the future the Linux kernel can be shrunk down to fit into the RAM of an ESP8266, I've seen some ARM noMMU targets with less 512KB RAM, but going down to 112KB is crazy. Maybe it's possible, but I think it's highly unlikely that somebody will put the effort in. Definitely not possible without willingness to patch the entire kernel.)

raspiduino, 2022-12-03:
> But [that's](http://dmitry.gr/index.php?r=05.Projects&proj=07.%20Linux%20on%208bit) [been](https://github.com/cnlohr/mini-rv32ima) [done](https://github.com/juiceRv/JuiceVm) before(multiple links!).

Yes, I'm porting uARM to esp8266 :))

<hr>

Btw I know this is xtensa linux repo, but do you have idea on how to build linux arm for noMMU target? Is it generally buildable from upstream kernel, or it needs special patches?

I want to port Linux to run natively on Nokia (Mediatek ARM SoC). I have already ported uARM to run on it, but it's not really fast due to swapping. Nokia 225 has 8MB RAM, which seems possible

Also I got source code for the OS running on it :))

jcmvbkbc, 2022-12-03:
> bFLT is somewhat [broken](https://github.com/jcmvbkbc/linux-xtensa/commit/0eb21ca4a4f22160b25ccdfd884260430b036ee2)(or I'm too stupid)

@max1220 you're correct, there is an issue in the elf2flt mainline, I've posted the following [pull request](https://github.com/uclinux-dev/elf2flt/pull/25) with a fix.

zap8600, 2022-12-06:
I was running scripts to build Linux, but I got an error when building Buildroot.
`  LINK    busybox_unstripped
Trying libraries: m
Failed: -Wl,--start-group -lm   -Wl,--end-group`
Did I skip a step?

zap8600, 2022-12-07:
@max1220 I've made an issue with the [xtensa-fun](https://github.com/max1220/xtensa-fun) repository. I'm unable to boot the image via QEMU. There are more details in the issue.

zap8600, 2022-12-07:
> I can't wait for proper FDPIC support(FDPIC are "real" ELFs and also should save some RAM, but it's a toolchain thing), because I think bFLT(Flat binaries, no/limited dynamic linking) is somewhat [...]

@max1220 what did you do to get to a shell? The kernel downloaded by xtensa-fun boot-loops.

zap8600, 2022-12-07:
I'm assuming that I need to find a mailing list to submit my problems.

zap8600, 2022-12-07:
> See http://wiki.osll.ru/doku.php/etc:users:jcmvbkbc:linux-xtensa:esp32 for the flashing tips.
> Also the kernel needs a compiler with -mforce-l32 support.
Since the kernel downloaded by xtensa-fun crashes due to an IRQ in the serial driver, I'm just going to use the [xtensa-6.0-esp32 branch of linux-xtensa](https://github.com/jcmvbkbc/linux-xtensa/tree/xtensa-6.0-esp32). Will the toolchain provided by Buildroot build the kernel? Do I just use `esp32_defconfig` for the kernel?

jcmvbkbc, 2022-12-07:
> I'm just going to use the [xtensa-6.0-esp32 branch of linux-xtensa](https://github.com/jcmvbkbc/linux-xtensa/tree/xtensa-6.0-esp32).

Better use https://github.com/jcmvbkbc/linux-xtensa/tree/xtensa-6.1-esp32

> Will the toolchain provided by Buildroot build the kernel?

It will if it includes -mforce-l32 support. You can add it by dropping the following [patch](https://github.com/jcmvbkbc/gcc-xtensa/commit/ab117bd279248aca6de8a590b3d433d50a0be9be.patch) into the buildroot's package/gcc/... for the chosen gcc version.

jcmvbkbc, 2022-12-07:
> Do I have to change anything in the buildroot config

No, no need to change anything.

zap8600, 2022-12-07:
> It will if it includes -mforce-l32 support. You can add it by dropping the following [patch](https://github.com/jcmvbkbc/gcc-xtensa/commit/ab117bd279248aca6de8a590b3d433d50a0be9be.patch) into the buildroot's package/gcc/... for the chosen gcc version.

I'm using gcc 11.3.0. Is there a specific way to patch it? Do I have to patch it manually?

zap8600, 2022-12-08:
> For gcc-11 and earlier the following patch should work: https://github.com/jcmvbkbc/gcc-xtensa/commit/021f95d99a68b87643fdc06b3f17da1f206f33b3.patch

This patch almost worked. I got these errors when the patch was automatically applied.

`Applying 021f95d99a68b87643fdc06b3f17da1f206f33b3.patch using patch: 
patching file gcc/config/xtensa/constraints.md
patching file gcc/config/xtensa/xtensa.c
Hunk #1 succeeded at 1864 (offset 63 lines).
Hunk #2 succeeded at 2388 (offset 102 lines).
Hunk #3 succeeded at 2399 (offset 102 lines).
Hunk #4 succeeded at 2550 (offset 103 lines).
patching file gcc/config/xtensa/xtensa.md
Hunk #1 FAILED at 532.
Hunk #2 FAILED at 569.
Hunk #3 succeeded at 840 (offset 43 lines).
Hunk #4 succeeded at 850 (offset 43 lines).
Hunk #5 succeeded at 870 (offset 43 lines).
Hunk #6 succeeded at 879 (offset 43 lines).
2 out of 6 hunks FAILED -- saving rejects to file gcc/config/xtensa/xtensa.md.rej
patching file gcc/config/xtensa/xtensa.opt
Hunk #1 FAILED at 45.
1 out of 1 hunk FAILED -- saving rejects to file gcc/config/xtensa/xtensa.opt.rej
make: *** [package/pkg-generic.mk:252: /home/zane/espLinux/xtensa-fun/buildroot/output/build/host-gcc-initial-11.3.0/.stamp_patched] Error 1
~/espLinux/xtensa-fun`

Do I need to choose an older gcc version? If so, then which one do I need to pick?

jcmvbkbc, 2022-12-08:
> This patch almost worked

Sigh ):

> Do I need to choose an older gcc version?

Could you instead use gcc-12?

zap8600, 2022-12-08:
> Could you instead use gcc-12?

I can use it. What patch do I need to use?

jcmvbkbc, 2022-12-08:
The one mentioned above: https://github.com/jcmvbkbc/gcc-xtensa/commit/ab117bd279248aca6de8a590b3d433d50a0be9be.patch

zap8600, 2022-12-08:
> The one mentioned above: https://github.com/jcmvbkbc/gcc-xtensa/commit/ab117bd279248aca6de8a590b3d433d50a0be9be.patch

The patch finally worked, but I got an error what installing kernel headers.

`>>> linux-headers xtensa-6.0-esp32 Installing to staging directory
(cd /home/zane/espLinux/xtensa-fun/buildroot/output/build/linux-headers-xtensa-6.0-esp32; PATH="/home/zane/espLinux/xtensa-fun/buildroot/output/host/bin:/home/zane/espLinux/xtensa-fun/buildroot/output/host/sbin:/home/zane/espLinux/xtensa-fun/esp-idf/components/esptool_py/esptool:/home/zane/espLinux/xtensa-fun/esp-idf/components/espcoredump:/home/zane/espLinux/xtensa-fun/esp-idf/components/partition_table:/home/zane/espLinux/xtensa-fun/esp-idf/components/app_update:/home/zane/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:/home/zane/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/zane/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/zane/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/home/zane/.espressif/python_env/idf5.1_py3.9_env/bin:/home/zane/espLinux/xtensa-fun/esp-idf/tools:/home/zane/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games:/home/zane/espLinux/xtensa-fun/buildroot/output/host/bin:/home/zane/espLinux/xtensa-fun/buildroot/output/host/xtensa-buildroot-uclinux-uclibc/bin" /usr/bin/make  ARCH=xtensa HOSTCC="/usr/bin/gcc" HOSTCFLAGS="" HOSTCXX="/usr/bin/g++" INSTALL_HDR_PATH=/home/zane/espLinux/xtensa-fun/buildroot/output/host/xtensa-buildroot-uclinux-uclibc/sysroot/usr headers_install)
make[1]: Entering directory '/home/zane/espLinux/xtensa-fun/buildroot/output/build/linux-headers-xtensa-6.0-esp32'
  INSTALL /home/zane/espLinux/xtensa-fun/buildroot/output/host/xtensa-buildroot-uclinux-uclibc/sysroot/usr/include
make[1]: Leaving directory '/home/zane/espLinux/xtensa-fun/buildroot/output/build/linux-headers-xtensa-6.0-esp32'
if ! support/scripts/check-kernel-headers.sh  /home/zane/espLinux/xtensa-fun/buildroot/output/build  /home/zane/espLinux/xtensa-fun/buildroot/output/host/xtensa-buildroot-uclinux-uclibc/sysroot  5.17 strict; then exit 1; fi
Incorrect selection of kernel headers: expected 5.17.x, got 6.0.x
make: *** [package/pkg-generic.mk:333: /home/zane/espLinux/xtensa-fun/buildroot/output/build/linux-headers-xtensa-6.0-esp32/.stamp_staging_installed] Error 1
~/espLinux/xtensa-fun`

I'm using the [xtensa-6.0-esp32](https://github.com/jcmvbkbc/linux-xtensa/tree/xtensa-6.0-esp32) to install headers. Do I need to change the repository branch or kernel header series in the buildroot config?

jcmvbkbc, 2022-12-08:
> I'm using the xtensa-6.0-esp32 to install headers.

There is absolutely no need to do that. Just use one of the standard variants offered by the buildroot.

zap8600, 2022-12-09:
> There is absolutely no need to do that. Just use one of the standard variants offered by the buildroot.

That fixed that issue. Now I have a new one. After a couple of hours building Linux, I was finally able to test it. Unfortunately, a got another error.

`========== ESP32 LINUX BOOTLOADER ==========                                    
[    0.000000] Ignoring boot parameters at (ptrval)                             
[    0.000000] Linux version 6.1.0-rc1-g3671f419d7d2 (linuxlite@linuxlite-Aspir2
[    0.000000] config ID: c2bcfffe:1cc5fe96                                     
[    0.000000] earlycon: esp32uart0 at MMIO32 0x3ff40000 (options '115200n8')   
[    0.000000] printk: bootconsole [esp32uart0] enabled                         
[    0.000000] **********************************************************       
[    0.000000] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **       
[    0.000000] **                                                      **       
[    0.000000] ** This system shows unhashed kernel memory addresses   **       
[    0.000000] ** via the console, logs, and other interfaces. This    **       
[    0.000000] ** might reduce the security of your system.            **       
[    0.000000] **                                                      **       
[    0.000000] ** If you see this message and you are not debugging    **       
[    0.000000] ** the kernel, report this immediately to your system   **       
[    0.000000] ** administrator!                                       **       
[    0.000000] **                                                      **       
[    0.000000] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **       
[    0.000000] **********************************************************       
ets Jul 29 2019 12:21:46                                                        
                                                                                
rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)                      
ets Jul 29 2019 12:21:46`

It just repeats the following below over and over.

`rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)                      
ets Jul 29 2019 12:21:46`

Now how do I fix it?

zap8600, 2022-12-09:
> That's in QEMU, right? I guess its timings don't match that of the real hardware.
> I've pushed the fix that just disables all watchdogs to the xtensa-6.1-esp32 branch.

If only I saw that you pushed the fix before I patched the kernel manually. Anyways, I finally reached the shell. Thanks for all of your help.

Here is some context on why I wanted to do this. Me and my friend are building a small PC using the ESP32 as the processor. We thought it would be funny if it ran Linux. Now that we know that running Linux on the ESP32 is possible, we can order the hardware, put it together, flash the image to the flash, and give it a go. 

Now that I think about it, there's a ESP32-S3 module that has 8MB os PSRAM and up to 32MB of flash. Could that run Linux? Would it also be possible to run RV32 noMMU Linux on the ESP32-C3? These are only a couple of things that could be something interesting. 

Anyways, thanks for the help and for the cool project.

jcmvbkbc, 2022-12-09:
> there's a ESP32-S3 module that has 8MB os PSRAM and up to 32MB of flash. Could that run Linux?

From its datasheet it looks like it should be able to. I guess I'll have a look.

zap8600, 2022-12-11:
> From its datasheet it looks like it should be able to. I guess I'll have a look.

It would be pretty cool if it could. The native USB support might add interesting possibilities to explore.

zap8600, 2023-01-07:
I guess I should close this now.

closed 2023-01-07 by zap8600