jcmvbkbc/linux-xtensa #5: ESP32-S3?

zap8600, 2023-01-08:
How do I run this on an ESP32-S3?

jcmvbkbc, 2023-01-08:
I guess the same way you run it on ESP32. You'd need to figure the address where the kernel goes and FLASH/PSRAM caches setup. Take [this branch](https://github.com/jcmvbkbc/linux-xtensa/commits/xtensa-6.1-esp32) for a template, add a core variant for ESP32-S3 at [this step](https://github.com/jcmvbkbc/linux-xtensa/commit/8444d421747475440a8ae7790eb7a2881e19af5b), add different DTS and change kernel config specifying correct core and addresses at [this step](https://github.com/jcmvbkbc/linux-xtensa/commit/a68faaed86e69cfef25b7a50a3ed54eca1082ef7) and use a compiler for the ESP32-S3 to build the kernel.

zap8600, 2023-01-11:
Where would I find the headers to add the ESP32-S3's core variant. After that, what would I need to change in the DTS. Then I need to know what the correct core and address are so I can change them.

jcmvbkbc, 2023-01-11:
> Where would I find the headers to add the ESP32-S3's core variant

I'd start [here](https://github.com/espressif/xtensa-overlays), but I see that these overlays are incomplete. Maybe ask for the linux parts? Or maybe the overlay was published elsewhere, I just couldn't find it by quick googling.

> what would I need to change in the DTS

Take the DST for the ESP32 and look at the devices and memories that you care about. Fix the register addresses/sizes that are wrong.

> what the correct core and address are

The correct core is esp32s3, or whatever you name it when you import it. The address -- you need to think about it: it must be in the FLASH, be mappable through the FLASH cache and you'd need to make a partition in the FLASH for it. It's not the single correct answer, but rather a range of more or less convenient options.

zap8600, 2023-01-11:
> I'd start [here](https://github.com/espressif/xtensa-overlays), but I see that these overlays are incomplete. Maybe ask for the linux parts? Or maybe the overlay was published elsewhere, I just couldn't find it by quick googling.

Ask Espressif for the Linux parts? What do I exactly need? What's missing?

zap8600, 2023-01-11:
@jcmvbkbc I think I found the [missing headers](https://github.com/espressif/esp-idf/tree/master/components%2Fxtensa%2Fesp32s3%2Finclude%2Fxtensa%2Fconfig).

zap8600, 2023-01-11:
I don't think I need to change anything else besides the core variant headers. I'll try to build it and give it a test in QEMU. Please tell me if I need to change anything else.

jcmvbkbc, 2023-01-11:
> I think I found the [missing headers](https://github.com/espressif/esp-idf/tree/master/components%2Fxtensa%2Fesp32s3%2Finclude%2Fxtensa%2Fconfig).

Yeah, that's correct.

> tell me if I need to change anything else.

For the linux kernel itself I think that's all.

zap8600, 2023-01-11:
> Yeah, that's correct.

> For the linux kernel itself I think that's all.
I don't need to change anything else at all? If so, then what about buildroot?

jcmvbkbc, 2023-01-11:
> I don't need to change anything else at all?

Check FLASH cache setup in the bootloader and rebuild it with the esp32s3 compiler.

> what about buildroot?

No changes, just rebuild with the esp32s3 compiler.

zap8600, 2023-01-11:
> > I don't need to change anything else at all?
> 
> Check FLASH cache setup in the bootloader and rebuild it with the esp32s3 compiler.
> 
> > what about buildroot?
> 
> No changes, just rebuild with the esp32s3 compiler.

So all I need to do is switch to the ESP32-S3 compiler?

jcmvbkbc, 2023-01-11:
> So all I need to do is switch to the ESP32-S3 compiler?

Yes.

zap8600, 2023-01-17:
There were problems when compiling Linux, so I don't think it'll work for now.

max1220, 2023-01-17:
I have compiled Linux for the ESP32 S3 successfully.
I don't think I've done no more than importing the missing headers, and adding a DTS.

Keep in mind that I've not tried booting it yet, because QEMU would need a port for the ESP32 as well(at least the correct memory mapping).
Also the bootloader currently uses ESP32-specific ROM functions, and so currently doesn't work on the ESP32 S3.

jcmvbkbc, 2023-01-18:
> There were problems when compiling Linux, so I don't think it'll work for now.

@zap8600 there shouldn't be any if things are configured correctly. I can take a look if you share the build command, build log and the source tree.

zap8600, 2023-01-18:
> I don't think I've done no more than importing the missing headers, and adding a DTS.

Can you provide the DTS?

max1220, 2023-01-24:
I'm currently trying to do the same.
I've gotten Linux to compile, and I've added a new DTS for the ESP32 S3.
There are some more differences between the ESP32 and the S3.
You'll at least need to adjust the hardcoded values in `arch/xtensa/platforms/esp32/setup.c` to setup the watchdog timer(I've commented out that section for the moment).

Here's the DTS I've used:
```
// SPDX-License-Identifier: GPL-2.0
/dts-v1/;

/ {
	compatible = "esp32";
	#address-cells = <1>;
	#size-cells = <1>;
	interrupt-parent = <&pic>;

	chosen {
		bootargs = "earlycon=esp32uart,mmio32,0x60000000,115200n8 console=ttyS0,115200n8 debug no_hash_pointers ";
	};

	aliases {
		serial0 = "/soc/serial@60000000";
	};

	memory@0 {
		device_type = "memory";
		reg = <0x3D000000 0x00200000>;
	};

	cpus {
		#address-cells = <1>;
		#size-cells = <0>;
		cpu@0 {
			compatible = "cdns,xtensa-cpu";
			reg = <0>;
			clocks = <&osc>;
		};
	};

	clocks {
		osc: clk160 {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <160000000>;
		};
		apb: clk80 {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <80000000>;
		};
	};

	pic: pic {
		compatible = "cdns,xtensa-pic";
		/* one cell: internal irq number,
		 * two cells: second cell == 0: internal irq number
		 *            second cell == 1: external irq number
		 */
		#interrupt-cells = <2>;
		interrupt-controller;
	};

	soc {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		interrupt-parent = <&intc>;
		ranges;

		intc: intc@600C2000 {
[... 178 lines cut ...]

static void IRAM_ATTR loop_forever() {
	while(1) { ;; }
}

void app_main()
{
	dbg_echo("\n========== ESP32 LINUX BOOTLOADER ==========\n");

	if (0) {
		// single-core mode
		map_flash(0);
		jump(flash_i_vaddr); // should never return
	} else {
		// dual-core mode
		map_flash(1);
		boot_app_cpu(flash_i_vaddr);
		dbg_echo("\n========== PRO CPU STILL RUNNING ==========\n");
		loop_forever();
	}
}
```

Unfortunately, I've had some problems with using both the USB JTAG with OpenOCD, and running a ESP32 S3 in QEMU(which doesn't support it yet, even after trying to hack in a compatible memory mapping and core headers, but maybe I'm not capable enough for this).
So debugging will be a little slower than possible, maybe.

zap8600, 2023-05-03:
I'm finally going to come back to this. So from what I understand so far, the bootloader needs to be rewritten because it uses functions that are only available on the ESP32, not the ESP32-S3. Is this correct?

jcmvbkbc, 2023-05-04:
> the bootloader needs to be rewritten because it uses functions that are only available on the ESP32, not the ESP32-S3

Yeah, it looks like it could use spi_flash_mmap or esp_partition_map. We're losing control over the chosen virtual address (anyway it will be the same for each boot), but getting portability. E.g. the following works for me on esp32:
```
        const void *ptr;
        spi_flash_mmap_handle_t handle;
        spi_flash_mmap(0x00040000, 0x001c0000, SPI_FLASH_MMAP_INST, &ptr, &handle);
        printf("ptr = %p\n", ptr);
        spi_flash_mmap(0x00200000, 0x00200000, SPI_FLASH_MMAP_INST, &ptr, &handle);
        printf("ptr = %p\n", ptr);
        spi_flash_mmap_dump();
```
However I see a different issue here (as well as with my original code): although the mapping succeeds for the whole 4MB - 256KB FLASH range, not all of it is actually readable. The kernel is accessible, but part of the rootfs is not ):

jcmvbkbc, 2023-05-04:
> the bootloader needs to be rewritten because it uses functions that are only available on the ESP32, not the ESP32-S3

Yeah, it looks like it could use spi_flash_mmap or esp_partition_map. We're losing control over the chosen virtual address (anyway it will be the same for each boot), but getting portability. E.g. the following works for me on esp32:
```
        const void *ptr;
        spi_flash_mmap_handle_t handle;
        spi_flash_mmap(0x00040000, 0x001c0000, SPI_FLASH_MMAP_INST, &ptr, &handle);
        printf("ptr = %p\n", ptr);
        spi_flash_mmap(0x00200000, 0x00200000, SPI_FLASH_MMAP_INST, &ptr, &handle);
        printf("ptr = %p\n", ptr);
        spi_flash_mmap_dump();
```
Or the same thing but with partitions:
```
static void IRAM_ATTR map_flash_and_go(void)
{
        const void *ptr0, *ptr;
        spi_flash_mmap_handle_t handle;
        esp_partition_iterator_t it;
        const esp_partition_t *part;

        it = esp_partition_find(ESP_PARTITION_TYPE_ANY, ESP_PARTITION_SUBTYPE_ANY, "linux");
        part = esp_partition_get(it);
        esp_partition_mmap(part, 0, part->size, SPI_FLASH_MMAP_INST, &ptr0, &handle);
        printf("ptr = %p\n", ptr0);

        it = esp_partition_find(ESP_PARTITION_TYPE_ANY, ESP_PARTITION_SUBTYPE_ANY, "rootfs");
        part = esp_partition_get(it);
        esp_partition_mmap(part, 0, part->size, SPI_FLASH_MMAP_INST, &ptr, &handle);
        printf("ptr = %p\n", ptr);

        spi_flash_mmap_dump();

        asm volatile ("jx %0" :: "r"(ptr0));
}
```
However I see a different issue here (as well as with my original code): although the mapping succeeds for the whole 4MB - 256KB FLASH range, not all of it is actually readable. The kernel is accessible, but part of the rootfs is not ):

jcmvbkbc, 2023-05-05:
> the same thing but with partitions

Somehow it doesn't work on esp32s3, `esp_partition_mmap` returns `ESP_ERR_NO_MEM`. However it works when FLASH is mapped as `SPI_FLASH_MMAP_DATA`. Looks like a bug to me, I'm playing with the following workaround:
```
diff --git a/components/soc/esp32s3/include/soc/ext_mem_defs.h b/components/soc/esp32s3/include/soc/ext_mem_defs.h
index 9bb1d2d3125a..b2d04f49c87f 100644
--- a/components/soc/esp32s3/include/soc/ext_mem_defs.h
+++ b/components/soc/esp32s3/include/soc/ext_mem_defs.h
@@ -46,7 +46,7 @@ extern "C" {
 
 //TODO, remove these cache function dependencies
 #define CACHE_IROM_MMU_START            0
-#define CACHE_IROM_MMU_END              Cache_Get_IROM_MMU_End()
+#define CACHE_IROM_MMU_END              (0x1ff * 4)
 #define CACHE_IROM_MMU_SIZE             (CACHE_IROM_MMU_END - CACHE_IROM_MMU_START)
 
 #define CACHE_DROM_MMU_START            CACHE_IROM_MMU_END
```

max1220, 2023-05-05:
@zap8600 You're right, AFAIK there is currently no working boot loader for the ESP32 S3. I've been busy with other things(and was somewhat held back by brocken openOCD, having the wrong board with too little RAM).

You also probably need to modify some memory-mapped IO addresses(e.g. UART, inerrupt controller, watchdog).

I having static address mappings would be great, since new addresses means recompiling the kernel every time the bootloader is built, and kernels aren't portable between bootloader versions. 

@jcmvbkbc 
I don't think that's a bug; I think the implementation tries to map the required address space as separate "pages", and since there are only so many "page table entries", and FreeRTOS already allocated some, there are not enough left(AFAIK the ESP32 S3 only has enough "pages" on the IROM to map IROM once).
I also don't know if the entire mapped region is "contigous".
Also make sure to disable all the relevant " safety checks" in the esp-idf(memory protections, watchdog timers, etc.).

I think the solution is writing a bootloader that reads, then re-writes the complete page table to map the entire address range at once.

To successfully do that, a working version of OpenOCD would be helpful, as well as QEMU for the ESP32 S3.

jcmvbkbc, 2023-05-05:
The workaround I mentioned [here](https://github.com/jcmvbkbc/linux-xtensa/issues/5#issuecomment-1535924297) is working, the FLASH partitions get mapped. The code is in the https://github.com/jcmvbkbc/esp-idf/tree/linux-5.0.1

zap8600, 2023-05-08:
> The workaround I mentioned [here](https://github.com/jcmvbkbc/linux-xtensa/issues/5#issuecomment-1535924297) is working, the FLASH partitions get mapped. The code is in the https://github.com/jcmvbkbc/esp-idf/tree/linux-5.0.1
> 
> And with that bootloader I was able to boot [this](https://github.com/jcmvbkbc/linux-xtensa/commits/xtensa-6.4-esp32) kernel on esp32s3 with 8MB PSRAM/8MB FLASH up to the switch to userspace.

I'll look for your ESP32-S3 code in that repo and modify my scripts to build it.

jcmvbkbc, 2023-05-08:
> I'll look for your ESP32-S3 code in that repo and modify my scripts to build it.

I've added a few notes on the whole process [here](http://wiki.osll.ru/doku.php/etc:users:jcmvbkbc:linux-xtensa:esp32s3)

zap8600, 2023-05-08:
> I've added a few notes on the whole process [here](http://wiki.osll.ru/doku.php/etc:users:jcmvbkbc:linux-xtensa:esp32s3)

I have a question. Are directories like `build-xtensa-fdpic-toolchain-esp32s3` and `build-xtensa-fdpic-nothread-esp32s3` created by the user?

jcmvbkbc, 2023-05-08:
> Are directories like build-xtensa-fdpic-toolchain-esp32s3 and build-xtensa-fdpic-nothread-esp32s3 created by the user?

Yes. I guess it's going to be easier if I just integrate toolchain building into the crosstool-NG, it makes things easier to reproduce, but slows down development a little.

max1220, 2023-05-09:
> I've added a few notes on the whole process [here](http://wiki.osll.ru/doku.php/etc:users:jcmvbkbc:linux-xtensa:esp32s3)

AWESOME! You're an absolute legend, kudos to you!
Your previous post is what got me started on this ESP32 Linux journey, and the reason I wrote my first kernel code.

Is this using the more efficient XIP-aware FDPIC that can map from cramfs?
My previous experiments all used bFLT, and the way worse memory usage that causes.

Just as an aside, since you're the expert, do you think it's plausible to run Linux and FreeRTOS "in parallel"(e.g. on separate cores?).
As far as I'm aware, the APP/PRO CPUs should have a separate enough memory mapping. If somebody/me implements such a hack, this would make using the hardware from Linux way easier(e.g. just do a "RPC" call to FreeRTOS to hardware stuff).
I'm specifically thinking about Wifi/Bluetooth which is using a binary blob driver for FreeRTOS making porting hard, and USB which seems underdocumented and hard to implement. I think the combination of a embedded RTOS with a full Linux could be very useful. 

Thank you for your time and awesome contributions!

zap8600, 2023-05-09:
> Is this using the more efficient XIP-aware FDPIC that can map from cramfs?
> My previous experiments all used bFLT, and the way worse memory usage that causes.

I do believe so. 

> As far as I'm aware, the APP/PRO CPUs should have a separate enough memory mapping. If somebody/me implements such a hack, this would make using the hardware from Linux way easier(e.g. just do a [...]

I'm somewhat familiar with using RPC systems, so I might be able to write a driver to help with this.

jcmvbkbc, 2023-05-09:
> Is this using the more efficient XIP-aware FDPIC that can map from cramfs?

Yes, the code is now truly shared between the processes and it can run directly from FLASH.
A few things are missing at the moment in the FDPIC support: there's no TLS and hence no NPTL, no -mlongcalls and hence limited distance between the functions that need to call each other, no stack unwinding and thus no C++, and its only hand-tested, i.e. I didn't run the compiler testsuite, so I don't really know what else might be broken.

> do you think it's plausible to run Linux and FreeRTOS "in parallel"(e.g. on separate cores?)

It should be. But since the cores share some hardware (e.g. the interrupt matrix) the IDF code may have issues with that and need some modifications.

zap8600, 2023-05-09:
What if we use the binary blobs for WiFi and BT drivers for Linux?

zap8600, 2023-05-10:
At the moment, I can only find the [BT stack for the ESP32](https://github.com/espressif/esp32-bt-lib), not the ESP32-S3, so we can only make drivers for the ESP32 for now.

ESP32DE, 2023-05-12:
just in time it plays with me [at this point](https://github.com/jcmvbkbc/gcc-xtensa/blob/fcc43d064e25ad2e17239d479552627f56a49a4c/gcc/config/xtensa/xtensa.h#L773-L774) 

```
 ../.././gcc/tm.h:24,
                  /home/dev23/s3/tensilica/gcc/gcc/libgcc/libgcc2.c:29:
/home/dev23/s3/tensilica/gcc/gcc/libgcc/../gcc/config/xtensa/xtensa.h:775:2: Fehler: #error Unsupported ABI
  775 | #error Unsupported ABI
      |  ^~~~~

```

```

/* Xtensa constant pool breaks the devices in crtstuff.c to control

   section in where code resides.  We have to write it as asm code.  Use

   a MOVI and let the assembler relax it -- for the .init and .fini

   sections, the assembler knows to put the literal in the right

   place.  */

#ifdef __FDPIC__

#if defined(__XTENSA_WINDOWED_ABI__)

/// uuh : breaking here 

#error Unsupported ABI

#elif defined(__XTENSA_CALL0_ABI__)

#define CRT_CALL_STATIC_FUNCTION(SECTION_OP, FUNC) \

    asm (SECTION_OP "\n\

	movi\ta0, " USER_LABEL_PREFIX #FUNC "@GOTOFFFUNCDESC\n\

	add\ta0, a0, a12\n\

	l32i\ta0, a0, 0\n\

	mov\ta11, a12\n\

	callx0\ta0\n" \

	TEXT_SECTION_ASM_OP);

#endif

#else

#if defined(__XTENSA_WINDOWED_ABI__)

#define CRT_CALL_STATIC_FUNCTION(SECTION_OP, FUNC) \

    asm (SECTION_OP "\n\

	movi\ta8, " USER_LABEL_PREFIX #FUNC "\n\

	callx8\ta8\n" \

	TEXT_SECTION_ASM_OP);

#elif defined(__XTENSA_CALL0_ABI__)

#define CRT_CALL_STATIC_FUNCTION(SECTION_OP, FUNC) \

    asm (SECTION_OP "\n\

	movi\ta0, " USER_LABEL_PREFIX #FUNC "\n\

	callx0\ta0\n" \

	TEXT_SECTION_ASM_OP);

#endif

#endif

```

What's missing here :)

jcmvbkbc, 2023-05-13:
> What's missing here :)

Right, I've fixed the esp32s3 overlay manually to use call0 ABI by default. I've pushed the overlay that I used to build the toolchain plugin [here](https://github.com/jcmvbkbc/config-esp32s3)

max1220, 2023-05-13:
Awesome! When I was building this yesterday, I had the same problem, and was trying some combination of the `-mabi` and `--with-abi=call0` flags in the `build-xtensa-fdpic-toolchain.sh`.

ESP32DE, 2023-05-13:
> > What's missing here :)
> 
> Right, I've fixed the esp32s3 overlay manually to use call0 ABI by default. I've pushed the overlay that I used to build the toolchain plugin [here](https://github.com/jcmvbkbc/config-esp32s3)

this solved the call0 ABI, thanks.

just in time build.sh ( LD )  plays with me [here](https://github.com/jcmvbkbc/binutils-gdb-xtensa/blob/0f7d842c396bce8e11834f7f8291739534a8a7c2/bfd/elf32-xtensa.c#L2683) 
```

  AS ldso/ldso/xtensa/resolve.os

  AR cr /home/dev23/s3/build-xtensa-fdpic-nothread-esp32s3/lib/libc.a

  STRIP -x -R .note -R .comment /home/dev23/s3/build-xtensa-fdpic-nothread-esp32s3/lib/libc.a

  AR cr /home/dev23/s3/build-xtensa-fdpic-nothread-esp32s3/lib/uclibc_nonshared.a

  STRIP -x -R .note -R .comment /home/dev23/s3/build-xtensa-fdpic-nothread-esp32s3/lib/uclibc_nonshared.a

  AR cr /home/dev23/s3/build-xtensa-fdpic-nothread-esp32s3/libc/libc_so.a

  STRIP -x -R .note -R .comment /home/dev23/s3/build-xtensa-fdpic-nothread-esp32s3/libc/libc_so.a

  AR cr /home/dev23/s3/build-xtensa-fdpic-nothread-esp32s3/lib/libpthread_nonshared.a

  STRIP -x -R .note -R .comment /home/dev23/s3/build-xtensa-fdpic-nothread-esp32s3/lib/libpthread_nonshared.a

echo "dev23debug_a"

dev23debug_a

  LD libuClibc-1.0.42.so

/home/dev23/s3/build-xtensa-fdpic-toolchain-esp32s3/root/lib/gcc/xtensa-linux-uclibcfdpic/14.0.0/../../../../xtensa-linux-uclibcfdpic/bin/ld: BFD (GNU Binutils) 2.40.50.20230424 assertion fail /home/dev23/s3/tensilica/binutils-gdb/binutils-gdb-xtensa/bfd/elf32-xtensa.c:2683

collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped

compilation terminated.

make: *** [libc/Makefile.in:78: /home/dev23/s3/build-xtensa-fdpic-nothread-esp32s3/lib/libc.so] Error 1

make: Leaving directory '/home/dev23/s3/uclibc-ng'

dev23@dev23-esp:~/s3$

```

```
 chunk = reloc_index / PLT_ENTRIES_PER_CHUNK;
  splt = elf_xtensa_get_plt_section (info, chunk);
  sgotplt = elf_xtensa_get_gotplt_section (info, chunk);
  BFD_ASSERT (splt != NULL && sgotplt != NULL);

  plt_base = splt->output_section->vma + splt->output_offset;
  got_base = sgotplt->output_section->vma + sgotplt->output_offset;

```

[libc/Makefile.in](https://github.com/jcmvbkbc/uclibc-ng-xtensa/blob/299d86dc999f0b6264cf46600829750cfe4a0a56/libc/Makefile.in#L76-L84) 

i did expanded with echo's for search my error
looks like this:

```

$(libc.depend): $(libc_OUT)/libc_so.a $(LIBS-libc.so)

	echo "dev23debug_a"

	$(call link.so,$(libc_FULL_NAME),$(ABI_VERSION))

	# link for backward compatibility

	echo "dev23debug_b"

	$(call link.so,$(libc_FULL_NAME),0)

	echo "dev23debug_c"

	$(Q)$(RM) $@

	echo "dev23debug_d"

	$(Q)cat $(top_srcdir)extra/scripts/format.lds > $@.tmp

	echo "dev23debug_e"

	$(Q)$(OUTPUT_FORMAT) >> $@.tmp

	echo "dev23debug_f"

	$(Q)echo "GROUP ( $(SHARED_LIBNAME) $(NONSHARED_LIBNAME) $(NONSHARED_LIBPTHREAD) $(ASNEEDED) )" >> $@.tmp

	echo "dev23debug_g"

	$(Q)mv $@.tmp $@

	echo "dev23debug_h"

```

i think, i have a config error in one code part.
anyone have an idea?

zap8600, 2023-05-13:
Would if be possible to give Linux access to **both** cores on the ESP32-S3?

jcmvbkbc, 2023-05-14:
> Would if be possible to give Linux access to both cores on the ESP32-S3?

I don't see significant issues with having SMP. One thing that must be done is implementation of different IPI mechanism and making IPI configurable.

jcmvbkbc, 2023-05-15:
> I guess it's going to be easier if I just integrate toolchain building into the crosstool-NG

This should work now, not perfectly, but it does the job. Updated [wiki](http://wiki.osll.ru/doku.php/etc:users:jcmvbkbc:linux-xtensa:esp32s3) with the details.

ESP32DE, 2023-05-15:
> > I guess it's going to be easier if I just integrate toolchain building into the crosstool-NG
> 
> This should work now, not perfectly, but it does the job. Updated [wiki](http://wiki.osll.ru/doku.php/etc:users:jcmvbkbc:linux-xtensa:esp32s3) with the details.

```
..
>>>   Generating filesystem image rootfs.cramfs
..
..
>>>   Generating filesystem image rootfs.tar
.. /home/dev23/s3/build-xtensa-2023.02-fdpic-esp32s3/staging

make: Leaving directory '/home/dev23/s3/buildroot'
..

```

@jcmvbkbc ..it does -**more** **more** **more**- -not only- the job - 
great work !!! 
now we can going on to fine tune the .config and step_by_step - great wiki update jcmvbkbc !!!
:)

for your reference: .. it's working! :)
[image]

ESP32DE, 2023-05-16:
@jcmvbkbc 

```

dev23@dev23-esp:~/s3/esp-idf/examples/get-started/linux_boot$ idf.py monitor

Executing action: monitor

Serial port /dev/ttyUSB0

Connecting...

Detecting chip type... ESP32-S3

Running idf_monitor in directory /home/dev23/s3/esp-idf/examples/get-started/linux_boot

Executing "/home/dev23/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/dev23/s3/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /home/dev23/s3/esp-idf/examples/get-started/linux_boot/build/linux_boot.elf -m '/home/dev23/.espressif/python_env/idf5.0_py3.10_env/bin/python' '/home/dev23/s3/esp-idf/tools/idf.py'"...

--- idf_monitor on /dev/ttyUSB0 115200 ---

--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3810,len:0x1098

load:0x403c9700,len:0xa24

load:0x403cc700,len:0x2d04

entry 0x403c988c

I (73) octal_psram: vendor id    : 0x0d (AP)

I (74) octal_psram: dev id       : 0x02 (generation 3)

I (74) octal_psram: density      : 0x03 (64 Mbit)

I (78) octal_psram: good-die     : 0x01 (Pass)

I (83) octal_psram: Latency      : 0x01 (Fixed)

I (89) octal_psram: VCC          : 0x01 (3V)

I (94) octal_psram: SRF          : 0x01 (Fast Refresh)

I (99) octal_psram: BurstType    : 0x01 (Hybrid Wrap)

I (105) octal_psram: BurstLen     : 0x01 (32 Byte)

I (111) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)

I (117) octal_psram: DriveStrength: 0x00 (1/1)

I (123) esp_psram: Found 8MB PSRAM device

I (127) esp_psram: Speed: 80MHz

I (130) cpu_start: Pro cpu up.

I (134) esp32s3-linux: eMbeddedHome-liosti-cpu1: 0h40375298

[... 572 lines cut ...]

	ptr = map_partition("rootfs", 0);
	printf("ptr rootfs = %p\n", ptr);

	spi_flash_mmap_dump();

	printf("Debugline_BG1\n");
	// asm volatile ("jx %0" :: "r"(ptr0));
	printf("Debugline_BG2\n");

}

```

..asm volatile ("jx %0" :: "r"(ptr0));.. should start the linux image right?
any special use in menuconfig what is not documented?
should i comment out the line now? 
ready for the boot...
:)

btw my boot load addr 1st is little different to yours, which hardware you use / bootloader ?
load:0x3fce3810,len:0x1098   <<<<
load:0x403c9700,len:0xa24
load:0x403cc700,len:0x2d04
entry 0x403c988c

ESP32DE, 2023-05-16:
all ptr addr:

// spi_flash_mmap_dump(); was comment out in fw//

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1098
load:0x403c9700,len:0xa24
load:0x403cc700,len:0x2d04
entry 0x403c988c
I (73) octal_psram: vendor id    : 0x0d (AP)
I (74) octal_psram: dev id       : 0x02 (generation 3)
I (74) octal_psram: density      : 0x03 (64 Mbit)
I (78) octal_psram: good-die     : 0x01 (Pass)
I (83) octal_psram: Latency      : 0x01 (Fixed)
I (89) octal_psram: VCC          : 0x01 (3V)
I (94) octal_psram: SRF          : 0x01 (Fast Refresh)
I (99) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (105) octal_psram: BurstLen     : 0x01 (32 Byte)
I (111) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (117) octal_psram: DriveStrength: 0x00 (1/1)
I (123) esp_psram: Found 8MB PSRAM device
I (127) esp_psram: Speed: 80MHz
I (130) cpu_start: Pro cpu up.
I (134) esp32s3-linux: eMbeddedHome-liosti-cpu1: 0h40375298
I (0) cpu_start: App cpu up.
I (593) esp_psram: SPI SRAM memory test OK
I (602) cpu_start: Pro cpu start user code
I (602) cpu_start: cpu freq: 160000000 Hz
I (602) cpu_start: Application information:
I (605) cpu_start: Project name:     linux_boot
I (610) cpu_start: App version:      v5.0.1-3-g8a29e7a6e1-dirty
I (617) cpu_start: Compile time:     May 16 2023 12:12:14
I (623) cpu_start: ELF file SHA256:  8a7e89243613caa6...
I (629) cpu_start: ESP-IDF:          v5.0.1-3-g8a29e7a6e1-dirty
I (636) cpu_start: Min chip rev:     v0.0
I (640) cpu_start: Max chip rev:     v0.99 
I (645) cpu_start: Chip rev:         v0.0
I (650) heap_init: Initializing. RAM available for dynamic allocation:
I (657) heap_init: At 3FC959A8 len 00053D68 (335 KiB): D/IRAM
I (664) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (670) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (676) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (683) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (691) spi_flash: detected chip: generic
I (695) spi_flash: flash io: dio
I (700) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (720) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
42830000..4283ffff: 5a8f331a
42840000..4284ffff: e32fc9f5
42850000..4285ffff: 4db92328
42860000..4286ffff: a2a14904
42870000..4287ffff: c040fe22
42880000..4288ffff: 7cdf6dc7
42890000..4289ffff: ff1b94cf
428a0000..428affff: df068c29
428b0000..428bffff: 9a51fe0b
428c0000..428cffff: a8a73b1a
428d0000..428dffff: a1d0cea5
428e0000..428effff: d525d875
428f0000..428fffff: 99a250d8
42900000..4290ffff: 5d802e8a
42910000..4291ffff: 360735ae
42920000..4292ffff: 56afb77b
42930000..4293ffff: 1f937139
[... 337 lines cut ...]
page 230: refcnt=1 paddr=107
page 231: refcnt=1 paddr=108
page 232: refcnt=1 paddr=109
page 233: refcnt=1 paddr=110
page 234: refcnt=1 paddr=111
page 235: refcnt=1 paddr=112
page 236: refcnt=1 paddr=113
page 237: refcnt=1 paddr=114
page 238: refcnt=1 paddr=115
page 239: refcnt=1 paddr=116
page 240: refcnt=1 paddr=117
page 241: refcnt=1 paddr=118
page 242: refcnt=1 paddr=119
page 243: refcnt=1 paddr=120
page 244: refcnt=1 paddr=121
page 245: refcnt=1 paddr=122
page 246: refcnt=1 paddr=123
page 247: refcnt=1 paddr=124
page 248: refcnt=1 paddr=125
page 249: refcnt=1 paddr=126
page 250: refcnt=1 paddr=127
Debugline_BG1
Debugline_BG2

```

jcmvbkbc, 2023-05-16:
> my boot load addr 1st is little different to yours, which hardware you use / bootloader ?

Esp32-S3-Devkitc-1N8R8, nothing special about the bootloader, just what was built with the SDK. The address is likely different because the revision 680509ab40d1 (from the boot log) was one revision ahead the github branch linux-5.0.1. It should be on github now.

ESP32DE, 2023-05-17:
...memememe..

[image]

Sigh :)

@jcmvbkbc i am not sure does my created toolchain work corectly / does the buildroot work right / does my used HW is ok, 
i get reboots for first try..

so if possible on your time bank :) , can you say/check : 

in buildroot config 

- the MMU is preset - is this right also for the esp32s3 buildroot?
- the [Header Src is 5.1](https://github.com/jcmvbkbc/buildroot/blob/9297e1add0533379cfe62b88ad6a62ef457745d5/configs/qemu_xtensa_kc705_xip_defconfig#LL8C1-L8C1) ( i get an error - toolchain is 6.1 so i set it up 6.1.x or higher ) 

- target files -[ cramfs and xip](https://github.com/jcmvbkbc/buildroot/blob/9297e1add0533379cfe62b88ad6a62ef457745d5/configs/qemu_xtensa_kc705_xip_defconfig#L11-L12) - i get a (linux)tar and (rootfs).cramfs 

not sure so i ask, the [overlay file](https://github.com/jcmvbkbc/buildroot/blob/9297e1add0533379cfe62b88ad6a62ef457745d5/configs/qemu_xtensa_kc705_xip_defconfig#L3) in buildroot where it comes from / should be? i searched but did not find in the folders, can this be? 

possible to provide your linux / root images for test the hardware here?
is it helpfull if i upload the toolchain as zip for check it out does it right compile?
also i can provide the linux / root images for your test.

Sigh :)

( Debug lines are only for the cpu's - does it start... for later to run things DUAL ....  )

jcmvbkbc, 2023-05-17:
> the MMU is preset - is this right also for the esp32s3 buildroot?

Yes, that's a temporary workaround, because the ELFs are not available for the noMMU. The busybox config has noMMU selected though.

> the [Header Src is 5.1](https://github.com/jcmvbkbc/buildroot/blob/9297e1add0533379cfe62b88ad6a62ef457745d5/configs/qemu_xtensa_kc705_xip_defconfig#LL8C1-L8C1) ( i get an error - toolchain is 6.1 so i set it up 6.1.x or higher )

Yes, you did the right thing fixing that, I had different headers versions for handcrafted toolchain and one built by ct-ng, will synchronize and update the buildroot config.

> target files -[ cramfs and xip](https://github.com/jcmvbkbc/buildroot/blob/9297e1add0533379cfe62b88ad6a62ef457745d5/configs/qemu_xtensa_kc705_xip_defconfig#L11-L12) - i get a (linux)tar and (rootfs).cramfs

Um... that buildroot config doesn't build the kernel. And now I realize that there are no instructions in the wiki for building the kernel image, like it's the most obvious thing. I obviously need to fix that one way or another.

> not sure so i ask, the [overlay file](https://github.com/jcmvbkbc/buildroot/blob/9297e1add0533379cfe62b88ad6a62ef457745d5/configs/qemu_xtensa_kc705_xip_defconfig#L3) in buildroot where it comes from / should be? i searched but did not find in the folders, can this be?

The good news with the gcc-13 and binutils-2.40 is that the toolchain may now be configured by the XTENSA_GNU_CONFIG environment variable pointing to a config-specific dynamic library, in that case no source code copying is necessary. I realize that buildroot would still apply the overlay, it really doesn't matter what's in it. I'll fix that part.

ESP32DE, 2023-05-17:
> Um... that buildroot config doesn't build the kernel. And now I realize that there are no instructions in the wiki for building the kernel image, like it's the most obvious thing. I obviously need [...]

lol :) ... i thinked the same - but was unsure :) but i thinked to start to build a linux kernel ( which is not preset )  itself which the cpu wants in first step for the rootfs loading :))) .. 

question: cramfs support must then be set in kernel build if we use cramfs as rootfs right? 

> The good news with the gcc-13 and binutils-2.40 is that the toolchain may now be configured by the XTENSA_GNU_CONFIG environment variable pointing to a config-specific dynamic library, in that case [...]

that a good thing !!! you are a legend @jcmvbkbc - and thank you for your fast reply !!!

:)

jcmvbkbc, 2023-05-17:
> possible to provide your linux / root images for test the hardware here?

http://jcmvbkbc.spb.ru/~dumb/tmp/202305161729/

jcmvbkbc, 2023-05-17:
> question: cramfs support must then be set in kernel build if we use cramfs as rootfs right?

Yes, it's enabled in the [esp32s3_defconfig](https://github.com/jcmvbkbc/linux-xtensa/blob/5d8354462a70a59efb4a64f118cbec2a64131e52/arch/xtensa/configs/esp32s3_defconfig#L65)

ESP32DE, 2023-05-17:
> > possible to provide your linux / root images for test the hardware here?
> 
> http://jcmvbkbc.spb.ru/~dumb/tmp/202305161729/

Dude !!! you did it!!
it's working and my hardware looks good :)
[Ubuntu 22.04-Klon (15-05-2023 )-screen0.webm](https://github.com/jcmvbkbc/linux-xtensa/assets/16070445/3b1bc094-dc77-4bed-8317-f3545117946e)

sry for fast and dirty webm for this great moment.
it goes 4k on day :)  Dude!! ( thank you! ) 
:)

ESP32DE, 2023-05-17:
@jcmvbkbc 
let's [get started](https://esp32.com/viewtopic.php?f=17&t=33675)

ESP32DE, 2023-05-19:
i try to make at end a makefile, 
to understand the steps, i go here very simplest - step by step - to final makefile with a simple hello world on the target

my "simplest" code:

```
/* helloworld.c */

#include <stdio.h>
#include <stdlib.h>
#include <string.h>

void main () {
  puts("Hello World\n");
  system("cat /proc/cpuinfo");
  exit(EXIT_SUCCESS);
}

```

at first:
export toolchain config and bin path

```
export XTENSA_GNU_CONFIG=/home/dev23/s3/xtensa-dynconfig/esp32s3.so
export PATH=/home/dev23/s3/crosstool-NG/builds/xtensa-esp32s3-linux-uclibcfdpic/bin:$PATH
```

compile:
```

dev23@dev23-esp:~/cross/prog_system_ps$ xtensa-linux-uclibcfdpic-gcc -c helloworld.c -o helloworld.o
dev23@dev23-esp:~/cross/prog_system_ps$ 
```

compile ok.

then (simple) link try ( without libc.a ) (1):

```
dev23@dev23-esp:~/cross/prog_system_ps$ xtensa-linux-uclibcfdpic-ld -o helloworld  helloworld.o -nostdlib 
xtensa-linux-uclibcfdpic-ld: warning: cannot find entry symbol _start; defaulting to 00400074
xtensa-linux-uclibcfdpic-ld: helloworld.o:(.literal+0x4): undefined reference to `puts'
xtensa-linux-uclibcfdpic-ld: helloworld.o:(.literal+0xc): undefined reference to `system'
dev23@dev23-esp:~/cross/prog_system_ps$ 

```
link messages cannot find entry symbol _start, simplest way to solve 
would be think in first step to add simple a _start symbol in the code:
also i can link with 

`--entry main`

i prefer the _start methode in code, 
is it the right way?

```
/*helloworld.c*/

#include <stdio.h>
#include <stdlib.h>
#include <string.h>

void main();

void  _start() {
 main();
}

void main () {
  puts("Hello World\n");
  system("cat /proc/cpuinfo");
  exit(EXIT_SUCCESS);
}

```

solved, but is it the right way?
do we need a return in the main or / and _start ( exit(EXIT_SUCCESS); ) ? 
do we can include a ld file for this _start provide for the linker to tell him where to start?
is there one in the toolchain build for this doing?
how to add on best way? ( i think export in the makefile ? ) 

```
dev23@dev23-esp:~/cross/prog_system_ps$ xtensa-linux-uclibcfdpic-ld -o helloworld  helloworld.o -nostdlib 
xtensa-linux-uclibcfdpic-ld: helloworld.o:(.literal+0x8): undefined reference to `puts'
xtensa-linux-uclibcfdpic-ld: helloworld.o:(.literal+0x10): undefined reference to `system'
dev23@dev23-esp:~/cross/prog_system_ps$ 

```

linker errors undefined reference puts & system,
i think this should be solved by add the libc.a from the host build right?
i am not sure does this be the right libc.a  also does this the right way to do this here, 
i think it should cause it is for the host to build the rootfs for the target
and our c archive.

i try:

`
/home/dev23/s3/build-xtensa-2023.02-fdpic-esp32s3/host/xtensa-buildroot-linux-uclibc/sysroot/usr/lib/libc.a
`
 
linker message is:

```
dev23@dev23-esp:~/cross/prog_system_ps$ xtensa-linux-uclibcfdpic-ld -o helloworld  helloworld.o /home/dev23/s3/build-xtensa-2023.02-fdpic-esp32s3/host/xtensa-buildroot-linux-uclibc/sysroot/usr/lib/libc.a -nostdlib  

xtensa-linux-uclibcfdpic-ld: /home/dev23/s3/build-xtensa-2023.02-fdpic-esp32s3/host/xtensa-buildroot-linux-uclibc/sysroot/usr/lib/libc.a(_fpmaxtostr.os): in function `_fpmaxtostr':

_fpmaxtostr.c:(.text+0x90): undefined reference to `__nedf2'

dev23@dev23-esp:~/cross/prog_system_ps$ 
```

do we need to include header folder by self / does the toolchain this do usually in this way? any export need like wise
`export LD_LIBRARY_PATH
`
where our tollchain lib path is.

Sigh :)

i am also on the path to add a "hello world package" to the buildroot just in time.
perhabs i found there a makefile how it is there done to get the symbols all right placed.
i am  not sure about the _start error message and what i should must do  - this would helpfull how it should be done on right way. 

thank you!

ESP32DE, 2023-05-19:
```
dev23@dev23-esp:~/cross/prog_system_ps$ xtensa-linux-uclibcfdpic-ld -A xtensa --abi-call0 -o helloworld  helloworld.o /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libc_pic.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/librt.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libpthread_nonshared.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libcrypt.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libpthread_nonshared_pic.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libdl_pic.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libcrypt_pic.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libresolv_pic.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libm_pic.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libnsl.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libc.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libresolv.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libnsl_pic.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libdl.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/libm.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/xtensa-linux-uclibcfdpic/sysroot/usr/lib/librt_pic.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/lib/bfd-plugins/libdep.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/lib/gcc/xtensa-linux-uclibcfdpic/14.0.0/libgcov.a /home/dev23/s3/xtensa-esp32s3-linux-uclibcfdpic/lib/gcc/xtensa-linux-uclibcfdpic/14.0.0/libgcc.a

dev23@dev23-esp:~/cross/prog_system_ps$

```

i took all i found ;) and this solve it but it looks soooo horrible
better way with shorts and exports.

not tested yet on the targe

:)

i

jcmvbkbc, 2023-05-19:
> then (simple) link try ( without libc.a ) (1):

But why? The whole point of having the toolchain with libc support is to use the libc. Just do

`xtensa-linux-uclibcfdpic-gcc helloworld.c -mfdpic -o helloworld`

or, if separate compile/build are required,

```
xtensa-linux-uclibcfdpic-gcc -c helloworld.c -mfdpic -o helloworld.o
xtensa-linux-uclibcfdpic-gcc helloworld.o -mfdpic -o helloworld
```
no need to do the compiler driver's job on your own.

ESP32DE, 2023-05-19:
> But why? The whole point of having the toolchain with libc support is to use the libc. Just do
>
> `xtensa-linux-uclibcfdpic-gcc helloworld.c -mfdpic -o helloworld`
>
> or, if separate compile/build are [...]

Dude 👍  asap try..

ESP32DE, 2023-05-19:
```
dev23@dev23-esp:~/cross/prog_system_ps$ ls
helloworld.c
dev23@dev23-esp:~/cross/prog_system_ps$ xtensa-linux-uclibcfdpic-gcc helloworld.c -mfdpic -o helloworld
dev23@dev23-esp:~/cross/prog_system_ps$ ls
helloworld  helloworld.c
dev23@dev23-esp:~/cross/prog_system_ps$ 

```

so simple :).. 
also removed the _start   not need
so an easy thing
lets check on the target..
( thank you !!! )

ESP32DE, 2023-05-19:
[image]

WoWZa!!!

Hello World is done!

thank you so much for your great service and your speed fast reply allways @jcmvbkbc 

:)

( i think the "Segmentation fault" comes from my Buildroot, which is not correctly, also i miss some network stuff on start ) 

[image]

ESP32DE, 2023-05-19:
what's next?
we go switching an Led  for the networkdriver ?

:)

jcmvbkbc, 2023-05-19:
> the "Segmentation fault"

I also observe it with the toolchain built by the ct-ng, but not with the hand-made toolchain. Quick investigation shows that it comes from an unexpected call to a dynamic linker and happens when libc is built with `-Os`, but doesn't happen when libc is built without optimizations. I'll take a deeper look.

jcmvbkbc, 2023-05-19:
> what's next?

That's up to you guys (: I see my role in making this _next_ part possible, there's a lot more to do here.

ESP32DE, 2023-05-19:
> I also observe it with the toolchain built by the ct-ng, but not with the hand-made toolchain. Quick investigation shows that it comes from an unexpected call to a dynamic linker and happens when [...]

FYI: i did notice that i can't ping local ip 127.0.0.0 with my (Buildroot) build but i can ping with your rootfs build.
I used the ct-ng toolchain in this case.

i will repeat the hello world also with my first toolchain build by hand,
i think if i use your fantastic step by step from the hello world
i am sure it will build also fine :)
Dude . so thankfull for your help here.

:)

ESP32DE, 2023-05-19:
> > what's next?
> 
> That's up to you guys (: I see my role in making this _next_ part possible, there's a lot more to do here.

let's do it! i am in.
i am looking at the wifi driver ... deeper now.
a good step will be to mount an sd card.
:)

[332 more replies]