jcmvbkbc/linux-xtensa #6: SMP and IPI for ESP32-S3

zap8600, 2023-07-03:
>> IPI trigger is not a part of the esp32 interrupt matrix and it doesn't have to be a part of an interrupt controller in the device tree, it may be a standalone device.

> So we need to find this device?

By 'device' I meant 'a node in the device tree'. The hardware itself is the system registers `SYSTEM_CPU_INTR_FROM_CPU_*_REG`.

I'd also suggest discussing it in a separate issue, as there's a lot of off-topic here already.

_Originally posted by @jcmvbkbc in https://github.com/jcmvbkbc/linux-xtensa/issues/5#issuecomment-1618998129_

zap8600, 2023-07-03:
So how can we find the node to add to the device tree? Are there any ESP-IDF functions we can use to find it?

jcmvbkbc, 2023-07-03:
> So how can we find the node to add to the device tree?

I'm not sure I understand the question. We need to just literally add a node to the esp32s3.dts. It should have the `compatible` property that would match a device driver that must be written to control it and the 'reg' property that would point to the right registers.

zap8600, 2023-07-03:
> I'm not sure I understand the question. We need to just literally add a node to the esp32s3.dts. It should have the `compatible` property that would match a device driver that must be written to control it and the `reg` property that would point to the right registers.

What sort of node are we looking for? Can we use ESP-IDF functions to find the node and its details? Is there a DTS in the ESP-IDF we could look at? Where in the DTS would we put the node?

jcmvbkbc, 2023-07-03:
> Can we use ESP-IDF functions to find the node and its details?

I don't think so.

> Is there a DTS in the ESP-IDF we could look at? 

No.

> What sort of node are we looking for? Where in the DTS would we put the node?

I'd suggest starting with the [specification](https://www.devicetree.org/specifications/) and then looking at other dts files in the linux kernel tree.

zap8600, 2023-07-04:
> I'd suggest starting with the [specification](https://www.devicetree.org/specifications/) and then looking at other dts files in the linux kernel tree.

Here is what I think is right so far.
```
cpu_0 {
    reg = <0x600C0000 0x30>;
};
```
I was unsure what to name it. I believe this would go under `soc` in the DTS. I still need to find what `compatible` should be. Anything else I need to do?

jcmvbkbc, 2023-07-04:
> reg = <0x600C0000 0x30>

That covers some unrelated registers and doesn't cover any IPI registers. Perhaps it should be split into multiple regions.

> I believe this would go under soc in the DTS.

Yes, that sounds right.

> I still need to find what compatible should be

It needs to be specific enough to indicate what hardware it is for, but otherwise it's up to you.

> Anything else I need to do?

The driver that would service this device and interact with the SMP guts of the xtensa linux kernel.

zap8600, 2023-07-04:
> That covers some unrelated registers and doesn't cover any IPI registers. Perhaps it should be split into multiple regions.

My bad. I thought it was what I should put for `SYSTEM_CPU_INTR_FROM_CPU_0_REG`.
> It needs to be specific enough to indicate what hardware it is for, but otherwise it's up to you.

Would something like `compatible = "esp,esp32s3-ipi";` work?

jcmvbkbc, 2023-07-04:
> I thought it was what I should put for SYSTEM_CPU_INTR_FROM_CPU_0_REG.

The format of `reg` property is `<base_address_cells size_cells …>`, with more base/size pairs in the place of … if necessary to represent multiple regions.

> Would something like compatible = "esp,esp32s3-ipi"; work?

Yes.

zap8600, 2023-07-04:
> The format of `reg` property is `<base_address_cells size_cells …>`, with more base/size pairs in the place of … if necessary to represent multiple regions. E.g. to cover the `SYSTEM_CPU_INTR_FROM_CPU_0_REG` alone it would be `<0x600c0030 4>`

So would this work?
```
ipi: ipi@600c0000 {
    compatible = "esp,esp32s3-ipi";
    cpu_0 {
        reg = <0x34 0x38>;
    };
    cpu_1 {
        reg = <0x38 0x3c>;
    };
};
```

jcmvbkbc, 2023-07-04:
> So would this work?

Nope. It has wrong sizes and it implies certain address translation which is not set up here. Please read the DTS spec for the details.
Again, to cover the `SYSTEM_CPU_INTR_FROM_CPU_0_REG` alone the reg would be `<0x600c0030 4>`.

zap8600, 2023-07-04:
> Nope. It has wrong sizes and it implies certain address translation which is not set up here. Please read the DTS spec for the details. Again, to cover the `SYSTEM_CPU_INTR_FROM_CPU_0_REG` alone the reg would be `<0x600c0030 4>`.

How about this?
```
ipi: ipi@600c0000 {
    compatible = "esp,esp32s3-ipi";
    cpu0@30 {
        reg = <0x0 0x4>;
    };
    cpu1@34 {
        reg = <0x34 0x4>;
    };
};

jcmvbkbc, 2023-07-04:
> How about this?

How about just building your kernel with this dts and checking how it works?

zap8600, 2023-07-04:
> How about just building your kernel with this dts and checking how it works?

Do I need to add another `cpu` to `cpus` first?

jcmvbkbc, 2023-07-04:
> Do I need to add another cpu to cpus first?

It doesn't matter: there's no code that would interpret this information.

zap8600, 2023-07-04:
> It doesn't matter: there's no code that would interpret this information.

Alright. Will the build fail if my DTS is invalid?

jcmvbkbc, 2023-07-05:
> Will the build fail if my DTS is invalid?

Depends on how invalid. Syntactically invalid dts will break the build, yes. Semantically inconsistent may cause dtc to print warnings. Otherwise you'll see issues at runtime.

zap8600, 2023-07-11:
I'm going to delay this for now. I'm waiting on some hardware to arrive. I have an off-topic question though. Would it be possible to wire an SD Card to the board, and then give Linux access to it? I just want to know if it's possible. If it is then I'll look into implementing it later.

jcmvbkbc, 2023-07-11:
> Would it be possible to wire an SD Card to the board (over either SDIO or SPI), and then give Linux access to it?

It should be. I don't see why it shouldn't be possible.

zap8600, 2023-07-12:
I'm going to close this for now. I think I should start with a less complex part of the chip to implement, like GPIO or SPI. I'll create an issue for questions on how I should implement these once I get to it.

closed 2023-07-12 by zap8600