jcmvbkbc/linux-xtensa #8: SPI Implementation

zap8600, 2023-07-17:
I'm going to implement an SPI driver for the S3. I'm referencing this specific post because there is an SPI driver that uses GPIO instead of the SPI hardware. This could allow for things like an SD card to be mounted.

zap8600, 2023-07-17:
I believe that @ESP32DE implemented the GPIO driver for Linux, so I'll find that and use that as a base. I'll then locate the pins required for SPI and add them to the DTS. I'll then add the SPI GPIO driver to the config, build and flash the new kernel, connect the board to an SPI device, and use a custom program to test it.

ESP32DE, 2023-07-17:
> I believe that @ESP32DE implemented the GPIO driver for Linux, so I'll find that and use that as a base. I'll then locate the pins required for SPI and add them to the DTS. I'll then add the SPI [...]

https://github.com/jcmvbkbc/linux-xtensa/issues/5#issuecomment-1566463889

zap8600, 2023-07-23:
Alright. I think I have a good enough understanding of the Linux kernel's build system to add the GPIO driver to it. However, I'm not sure how I should go about adding the new definitions to the DTS. How should I do this?

jcmvbkbc, 2023-07-23:
> how I should go about adding the new definitions to the DTS.

You need to understand what information about the hardware your driver needs and then either find an established technique to pass this kind of information (like a `reg` property for a set of memory regions)  or invent your own format. Then just put new node into the dts used in the build of your kernel.

zap8600, 2023-07-24:
Does this look good for a new `.dtsi` file?
```
// SPDX-License-Identifier: GPL-2.0
/dts-v1/;

/ {
	compatible = "esp32s3";
	#address-cells = <1>;
	#size-cells = <1>;
	interrupt-parent = <&pic>;

	chosen {
		bootargs = "earlycon=esp32s3uart,mmio32,0x60000000,115200n8,40000000 console=ttyS0,115200n8 debug rw root=mtd:data no_hash_pointers ";
	};

	aliases {
		serial0 = "/soc/serial@60000000";
	};

	cpus {
		#address-cells = <1>;
		#size-cells = <0>;
		cpu@0 {
			compatible = "cdns,xtensa-cpu";
			reg = <0>;
			clocks = <&osc>;
		};
	};

	clocks {
		osc: clk160 {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <160000000>;
		};
		apb: clk80 {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <80000000>;
		};
		serial_clk: serial_clk {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <40000000>;
		};
	};

	pic: pic {
		compatible = "cdns,xtensa-pic";
		/* one cell: internal irq number,
		 * two cells: second cell == 0: internal irq number
		 *            second cell == 1: external irq number
		 */
		#interrupt-cells = <2>;
		interrupt-controller;
	};

	soc {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		interrupt-parent = <&intc>;
		ranges;

		intc: intc@600c2000 {
			compatible = "esp,esp32-intc";
			reg = <0x600c2000 0x800>;
			/* first cell: interrupt matrix row
			 * second cell: internal xtensa core IRQ number to connect to
			 * third cell: target CPU
			 */
			#interrupt-cells = <3>;
			interrupt-controller;
			interrupt-parent = <&pic>;
		};

		serial0: serial@60000000 {
			device_type = "serial";
			compatible = "esp,esp32s3-uart";
			reg = <0x60000000 0x80>;
			interrupts = <27 1 0>; /* UART0, external IRQ1, CPU0 */
			clocks = <&serial_clk>;
		};

		lpm: lpm@60008000 {
			compatible = "syscon";
			reg = <0x60008000 0x1000>;
		};

		syscon@60008000 {
			compatible = "syscon-reboot";
			regmap = <&lpm>;
			offset = <0>;
			value = <0x80000000>;
			mask = <0x80000000>;
		};
        
		gpio-controller@60004000 {
			compatible = "gpio";
			reg = <0x60004000 0x1000>;
			gpio-controller;
			#gpio-cells = <2>;
			interrupt-controller;
			#interrupt-cells = <2>;
		};
		
		fragment@0 {
			target = <&gpio>;
			__overlay__ {
				gpio_pins: gpio_pins {
					gpio0: gpio@0 {
						compatible = "esp,esp32s3-gpio";
						reg = <0x60004000 0x0 0x4>;
						label = "GPIO0";
					};
					gpio1: gpio@1 {
						compatible = "esp,esp32s3-gpio";
						reg = <0x60004000 0x4 0x4>;
						label = "GPIO1";
					};
					// specify more GPIO pins here ...
				};
			};
		};
	};
};
```

jcmvbkbc, 2023-07-24:
> Does this look good for a new .dtsi file?

It doesn't, for the following reasons:
- there's no `gpio` phandle that it tries to apply an overlay to
- there's no need to do an overlay in the dts file, just write the node that you need

You may find the following documents useful:
https://docs.kernel.org/devicetree/bindings/submitting-patches.html
the following files in the linux source tree:
- Documentation/devicetree/bindings/gpio/gpio.txt
- Documentation/devicetree/bindings/pinctrl/pinctrl-bindings.txt
- Documentation/devicetree/bindings/spi/spi-gpio.yaml

zap8600, 2023-07-24:
> It doesn't, for the following reasons:
>
> * there's no `gpio` phandle that it tries to apply an overlay to
> * there's no need to do an overlay in the dts file, just write the node that you need
>
> You [...]

Alright. I think this will work with the driver now.
```
​// SPDX-License-Identifier: GPL-2.0
/dts-v1/;

/ {
	compatible = "esp32s3";
	#address-cells = <1>;
	#size-cells = <1>;
	interrupt-parent = <&pic>;

	chosen {
		bootargs = "earlycon=esp32s3uart,mmio32,0x60000000,115200n8,40000000 console=ttyS0,115200n8 debug rw root=mtd:data no_hash_pointers ";
	};

	aliases {
		serial0 = "/soc/serial@60000000";
	};

	cpus {
		#address-cells = <1>;
		#size-cells = <0>;
		cpu@0 {
			compatible = "cdns,xtensa-cpu";
			reg = <0>;
			clocks = <&osc>;
		};
	};

	clocks {
		osc: clk160 {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <160000000>;
		};
		apb: clk80 {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <80000000>;
		};
		serial_clk: serial_clk {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <40000000>;
		};
	};

	pic: pic {
		compatible = "cdns,xtensa-pic";
		/* one cell: internal irq number,
		 * two cells: second cell == 0: internal irq number
		 *            second cell == 1: external irq number
		 */
		#interrupt-cells = <2>;
		interrupt-controller;
	};

	soc {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		interrupt-parent = <&intc>;
		ranges;

		intc: intc@600c2000 {
			compatible = "esp,esp32-intc";
			reg = <0x600c2000 0x800>;
			/* first cell: interrupt matrix row
			 * second cell: internal xtensa core IRQ number to connect to
			 * third cell: target CPU
			 */
			#interrupt-cells = <3>;
			interrupt-controller;
			interrupt-parent = <&pic>;
		};

		serial0: serial@60000000 {
			device_type = "serial";
			compatible = "esp,esp32s3-uart";
			reg = <0x60000000 0x80>;
			interrupts = <27 1 0>; /* UART0, external IRQ1, CPU0 */
			clocks = <&serial_clk>;
		};

		lpm: lpm@60008000 {
			compatible = "syscon";
			reg = <0x60008000 0x1000>;
		};

		syscon@60008000 {
			compatible = "syscon-reboot";
			regmap = <&lpm>;
			offset = <0>;
			value = <0x80000000>;
			mask = <0x80000000>;
		};
        
		gpio-controller@60004000 {
			compatible = "gpio";
			reg = <0x60004000 0x1000>;
			gpio-controller;
			#gpio-cells = <2>;
			interrupt-controller;
			#interrupt-cells = <2>;
		};
		
		gpio {
			gpio_pins: gpio_pins {
				gpio0: gpio@0 {
					compatible = "esp,esp32s3-gpio";
					reg = <0x60004000 0x0 0x4>;
					label = "GPIO0";
				};
				gpio1: gpio@1 {
					compatible = "esp,esp32s3-gpio";
					reg = <0x60004000 0x4 0x4>;
					label = "GPIO1";
				};
				// specify more GPIO pins here ...
			};
		};
	};
};
```

ESP32DE, 2023-08-12:
> Alright. I think this will work with the driver now.
>
> ```
> ​// SPDX-License-Identifier: GPL-2.0
> /dts-v1/;
>
> / {
> 	compatible = "esp32s3";
> 	#address-cells = <1>;
> 	#size-cells = <1>;
> 	interrupt-parent = [...]

@zap8600 
Do you get along? if you need help, write.

Here I go the way for x-fb and xserver (xterm). The S3 rootfs image has at least 11.5 MB for me. 16 MB flash exhausts it at the line and rootfs is at 0x00b00000 and is just about enough. I then created and mounted a rootfs to NFS Share, swapped out parts so that Flash Partion becomes freer again. 16MB should be enough for NFS mounts at the moment for me. As info.
i am on the sd card spi / net spi now, that an easyier part and i think its come true soon. do you think on use a spi display?
i am on [ST7789](https://github.com/ESP32DE/Boot-Linux-ESP32S3-Playground/tree/main/DTSi) at time , here is [datsheet](https://www.buydisplay.com/download/ic/ST7789.pdf)

zap8600, 2023-08-14:
> @zap8600
> Do you get along? if you need help, write.
>
> Here I go the way for x-fb and xserver (xterm). The S3 rootfs image has at least 11.5 MB for me. 16 MB flash exhausts it at the line and rootfs [...]

No, not yet. Tell me how `xterm` goes, though.

ESP32DE, 2023-08-14:
> No, not yet. Tell me how `xterm` goes, though.

it depends what you want do - 
do you want run xterm on the s3 by xorg on the s3 itself..connected by uart
do you want run xterm on the s3 by xorg on the s3 itself..connected by pc ssh client to the s3 ssh server 
do you want run xterm on the s3 by remotly run xapps from a server where you login.. from a uart connection with s3
do you want run xterm on the s3 by remotly run xapps from a server where you login.. from a ssh connection on s3

xterm on S3Linux connected by uart console is working so far..
[image]

..need little more compilation work and patches. after [summerbreak](https://twitter.com/eMbeddedHome/status/1689894673328844800) i come back to this theme.

ESP32DE, 2023-08-14:
> No, not yet. Tell me how `xterm` goes, though.

@zap8600 
it depends what you want do -
do you want run xterm on the s3 by xorg on the s3 itself..connected by uart
do you want run xterm on the s3 by xorg on the s3 itself..connected by pc ssh client to the s3 ssh server
do you want run xterm on the s3 by remotly run xapps from a server where you login.. from a uart connection with s3
do you want run xterm on the s3 by remotly run xapps from a server where you login.. from a ssh connection on s3

xterm on S3Linux connected by uart console is working so far..
[image]

..need little more compilation work and patches. after summerbreak i come back to this theme.

JFYI:
in (any) case, you need/should to Cutting the cord of Dropbear/Busybox dependency, there is no support for run ssh/sshd by -X just -W or i understand it wrong [image]

you need to patch the origin ssh client (MMU) to (NoMMU), this also need to do for local things for the origin xserver.
this need to install openssl and use this instead dropbear's embedded limited stuff which you can not all do what you want.
you need patch few (MMU) things to (NoMMU) apps on build process. sure, thus all will blow up your rootfs...
i am to the target to use also sd card / usb stick / NAND flash so - this cramfs flash will not my borderline in anycase to go more on this to test more.

the most important - you need to expand[ this tiny little work a bit little more](https://github.com/jcmvbkbc/buildroot/commit/1440c3a03cfd1fe104bf7578ba3474218006fdfd) to get the sigs working right like they should do. 

It's a nice approach, but it's not enough. the xserver get this wrong understand and you are not been happy with your result.

https://github.com/jcmvbkbc/linux-xtensa/assets/16070445/cda52252-bdcb-449e-a977-145963adc617

I'm not quite done with it yet, because Busybox always delivers some surprises that you don't expect and you have go back to start your work from begin. ( the patch is just for the "exit" not for the wrong sigs, so you need work little further to get this out:

[image]

i switched now to own debug mode - 
therefore, I will rewrite the parts for myself/usecase and use neat comprehensible individual sources and say goodbye to them. It also means that I will go my own usecase way for now. I don't like dependencies in the code that I can't understand or the origin is obfuscated and not visible, or is not entered or specified, 100 times in a script with other scripts that are reloaded are scripted, especially in SSL/TLS. they are good approaches and comfortable ones that take away the thinking, but here you should do it yourself. My 5ct opinion. 

Also the original wget is used, wget from busybox does not support https.. So reinstall most of the parts yourself.
this this means also ftp / ftpd and the other internet tools.  
 
have a nice summerbreak

zap8600, 2023-08-15:
> @zap8600
> it depends what you want do -
> do you want run xterm on the s3 by xorg on the s3 itself..connected by uart
> do you want run xterm on the s3 by xorg on the s3 itself..connected by pc ssh [...]

Well, since I have yet to try and get WiFi working yet, I'll just use it over UART. Mind telling me how to set it up?

zap8600, 2023-08-16:
I'm gonna take the xterm talk to the main issue.

jcmvbkbc, 2023-08-16:
> I'm gonna take the xterm talk to the main issue.

I'd suggest making a separate issue (because the #5 is filled with offtopic and is barely readable/searchable), perhaps even in a separate repository (because surely it's not related to the linux kernel).

zap8600, 2023-08-16:
I'll do that soon.

sam-m-1234, 2023-08-20:
@zap8600  @jcmvbkbc on the topic of SPI, do you think it would be possible to borrow some of the dts/dtsi content from the zephyr project files, which seems to have lots of peripherals already added for the ESP32S3? For example, my current project's build/zephyr.dts file has the below content. Since it already has the registers, what else would be required to port across, for example, the SPI2 periperal? 

/dts-v1/;

/ {
	#address-cells = < 0x1 >;
	#size-cells = < 0x1 >;
	model = "esp32s3_devkitm";
	compatible = "espressif,esp32s3";
	chosen {
		zephyr,canbus = &twai;
		zephyr,entropy = &trng0;
		zephyr,flash-controller = &flash;
		zephyr,sram = &sram0;
		zephyr,console = &uart0;
		zephyr,shell-uart = &uart0;
		zephyr,flash = &flash0;
	};
	aliases {
		i2c-0 = &i2c0;
		watchdog0 = &wdt0;
		uart-0 = &uart0;
		sw0 = &button0;
		int0 = &interrupt0;
	};
	soc {
		#address-cells = < 0x1 >;
		#size-cells = < 0x1 >;
		compatible = "simple-bus";
		ranges;
		sram0: memory@3fc88000 {
			compatible = "mmio-sram";
			reg = < 0x3fc88000 0x77fff >;
		};
		intc: interrupt-controller@600c2000 {
			#interrupt-cells = < 0x1 >;
			compatible = "espressif,esp32-intc";
			interrupt-controller;
			reg = < 0x600c2000 0x1000 >;
			status = "okay";
			phandle = < 0x1 >;
		};
		rtc: rtc@60021000 {
			compatible = "espressif,esp32-rtc";
			reg = < 0x60021000 0x2000 >;
			xtal-freq = < 0x2 >;
			#clock-cells = < 0x1 >;
			status = "okay";
			phandle = < 0x2 >;
			rtc_timer: rtc_timer {
				compatible = "espressif,esp32-rtc-timer";
				slow-clk-freq = < 0x249f0 >;
				interrupts = < 0x27 >;
				interrupt-parent = < &intc >;
				status = "okay";
			};
		};
		flash: flash-controller@60002000 {
			compatible = "espressif,esp32-flash-controller";
			reg = < 0x60002000 0x1000 >;
			#address-cells = < 0x1 >;
			#size-cells = < 0x1 >;
			flash0: flash@0 {
				compatible = "soc-nv-flash";
				reg = < 0x0 0x800000 >;
				erase-block-size = < 0x1000 >;
				write-block-size = < 0x4 >;
				status = "okay";
				partitions {
					compatible = "fixed-partitions";
[... 335 lines cut ...]
			group2 {
				pinmux = < 0x227fe7 >;
				output-low;
			};
		};
		twai_default: twai_default {
			phandle = < 0xa >;
			group1 {
				pinmux = < 0x3a7fc5 >, < 0xff9d06 >;
			};
		};
	};
	buttons {
		compatible = "gpio-keys";
		button0: button_0 {
			gpios = < &gpio0 0x0 0x11 >;
			label = "BOOT Button";
			zephyr,code = < 0xb >;
		};
		interrupt0: interrupt_0 {
			gpios = < &gpio0 0xe 0x20 >;
			label = "SPI Slave Data Ready Interrupt Pin";
		};
	};
};

jcmvbkbc, 2023-08-20:
> what else would be required to port across, for example, the SPI2 periperal?

The driver code for the device and its dependencies.

sam-m-1234, 2023-08-21:
Ok thanks, I'll have a go at re-using some of the zephyr device driver code and see If I can make any progress towards adding a new peripheral.

zap8600, 2023-08-21:
It looks possible. It seems to use some parts of zephyr, but those can probably be converted.

closed 2023-10-24 by zap8600