jcmvbkbc/linux-xtensa #9: Xorg on S3

zap8600, 2023-08-16:
As suggested, I've made a separate issue for this.

zap8600, 2023-08-16:
@ESP32DE (sorry for the mention) mind telling me how to install `xterm` on my S3?

ESP32DE, 2023-08-16:
> @ESP32DE (sorry for the mention) mind telling me how to install `xterm` on my S3?

for first rapid try, just install xterm, ( see also X "tools" like mouse, keyboard anything whats forwarded to x or receive by x)  from the target package, it looks like xorg is a depend ( must check by controll the menu entry by "help" 
it shows you left side then the depends .. ( in menuconfig ) 

reconfig it for your usecase ( Xmouse yes or no, Xkeyboard yes or no and so on ) strip it down to your flash size
can be that you will miss few tools later / or size up your rootfs size and install all. 
i am out here @zap8600 for a while sry.  ( summerbreak and other things )

zap8600, 2023-08-16:
I'll probably just use a keyboard. I'll need to add `xterm`, `xorg`, and `xkeyboard` right?

zap8600, 2023-08-17:
Hey @ESP32DE I'm installing `xterm`, `xorg`, and `xserver`. How can I forward them over UART or SSH?

closed 2023-09-12 by zap8600