jcmvbkbc/linux-xtensa #10: USB Serial output instead of UART

zap8600, 2023-08-17:
What would it take to switch from UART to USB serial?

jcmvbkbc, 2023-09-09:
USB serial of the USB Serial/JTAG Controller (chapter 33 in the esp32s3 TRM)? It looks like a very simple serial port on the esp32s3 side. I'd make a simple serial driver for it and specify it in the DTS.

jcmvbkbc, 2023-09-09:
[This](https://github.com/jcmvbkbc/linux-xtensa/commit/014e0425162f2ead8dc3d4101bbdb92f99768a47) driver, the following [changes](https://github.com/jcmvbkbc/linux-xtensa/commit/033d502d368b49a5360072865864574d6904cb2f) in the dts and the following kernel command line should do it: `earlycon=esp32s3acm,mmio32,0x60038000 console=ttyACM1 debug rw root=mtd:rootfs no_hash_pointers`

zap8600, 2023-09-10:
> [This](https://github.com/jcmvbkbc/linux-xtensa/commit/014e0425162f2ead8dc3d4101bbdb92f99768a47) driver, the following [...]

I have to modify the dts and enable the driver myself, right?

jcmvbkbc, 2023-09-10:
> I have to modify the dts and enable the driver myself, right?

It would probably be easier that way for now. I'll include it into the xtensa-esp32 kernel branch based on v6.6-rc1 once it's out, but that will also require bootloader update.

zap8600, 2023-09-11:
> It would probably be easier that way for now. I'll include it into the xtensa-esp32 kernel branch based on v6.6-rc1 once it's out, but that will also require bootloader update.

So, I also have to modify my bootloader?

jcmvbkbc, 2023-09-11:
> So, I also have to modify my bootloader?

Not to use this driver, no.

zap8600, 2023-09-11:
> Not to use this driver, no.

Do I need to change the repository version in buildroot to `xtensa-6.5-esp32-base`?

jcmvbkbc, 2023-09-11:
I don't think so. Why?

zap8600, 2023-09-11:
> I don't think so. Why?

The driver isn't found in `xtensa-6.5-esp32-base`.

jcmvbkbc, 2023-09-11:
Above I gave you the link to the commit. You can take the change and apply it to any kernel version that you have.

zap8600, 2023-09-11:
> Above I gave you the link to the commit. You can take the change and apply it to any kernel version that you have.

I'm just going to switch to `xtensa-6.5-esp32`.

zap8600, 2023-09-12:
@jcmvbkbc after installing the new kernel, the driver works. However, I am no longer able t connect to the WiFi. The board detects the network, but just can't connect. Here is the output:
```
~ # wpa_supplicant -i espsta0 -c /etc/wpa_supplicant.conf
Successfully initialized wpa_supplicant
espsta0: SME: Trying to authenticate with c4:41:1e:19:b8:a9 (SSID='This one.' freq=2417 MHz)
espsta0: SME: Authentication request to the driver failed
espsta0: SME: Trying to authenticate with c4:41:1e:19:b8:a9 (SSID='This one.' freq=2417 MHz)
espsta0: SME: Authentication request to the driver failed
espsta0: SME: Trying to authenticate with c4:41:1e:19:b8:a9 (SSID='This one.' freq=2417 MHz)
espsta0: SME: Authentication request to the driver failed
espsta0: SME: Trying to authenticate with c4:41:1e:19:b8:a9 (SSID='This one.' freq=2417 MHz)
espsta0: SME: Authentication request to the driver failed
espsta0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="This one." auth_failures=1 duration=10 reason=CONN_FAILED
^Cespsta0: CTRL-EVENT-DSCP-POLICY clear_all
espsta0: CTRL-EVENT-DSCP-POLICY clear_all
nl80211: deinit ifname=espsta0 disabled_11b_rates=0
espsta0: CTRL-EVENT-TERMINATING 
~ #
```

jcmvbkbc, 2023-09-12:
> Can I now use something like picocom on the S3 to read data over the UART port?

I think so. It shouldn't be different from the getty that is already used.

zap8600, 2023-09-12:
> I think so. It shouldn't be different from the getty that is already used.

Will UART have a serial terminal or ESP-IDF output on it?

jcmvbkbc, 2023-09-12:
> Will UART have a serial terminal

Depends on what you connect to it. getty would give you the normal login prompt and then the shell and everything.

> ESP-IDF output

Depends on the IDF settings. I see some in the default config.

zap8600, 2023-09-12:
I'm on ACM, so it shouldn't have a UART terminal. I can probably disable any output from ESP-IDF.

zap8600, 2023-09-12:
I'll close this issue now, as it is no longer needed.

closed 2023-09-12 by zap8600