jcmvbkbc/linux-xtensa #11: GPIO implementation

zap8600, 2023-09-12:
What would it take for a GPIO driver to be made? I can't find the code @ESP32DE made in the original issue so I'm creating a new issue for this.

jcmvbkbc, 2023-09-12:
> I can't find the code

I believe there was no kernel code, but a userspace program that was writing registers directly.

zap8600, 2023-09-12:
> I believe there was no kernel code, but a userspace program that was writing registers directly.

Do you still have the program?

jcmvbkbc, 2023-09-12:
> Do you still have the program?

It's `devmem`, part of busybox. See https://olimex.wordpress.com/2023/06/27/running-linux-with-kernel-6-3-on-open-source-hardware-board-with-esp32-s3/ for the use example.

zap8600, 2023-09-12:
> It's `devmem`, part of busybox. See https://olimex.wordpress.com/2023/06/27/running-linux-with-kernel-6-3-on-open-source-hardware-board-with-esp32-s3/ for the use example.

What about in C code? I plan on trying to write a driver if I can figure out how.

jcmvbkbc, 2023-09-12:
> What about in C code?

Depends on how fancy/portable you want your code. You can just write data directly to the desired physical address from the userspace if you don't care at all.

> I plan on trying to write a driver

Then it might be a good idea to take a look at other similar drivers.

zap8600, 2023-09-12:
> Then it might be a good idea to take a look at other similar drivers.

Are there any similar drivers?

jcmvbkbc, 2023-09-12:
A [whole bunch](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/gpio?h=v6.5).

zap8600, 2023-09-12:
Could we use [the MMIO GPIO driver](https://github.com/jcmvbkbc/linux-xtensa/blob/master/drivers/gpio/gpio-mmio.c)?

jcmvbkbc, 2023-09-12:
IDK. Perhaps. Try and see?

zap8600, 2023-09-12:
What is the minimum for writing my own driver?

jcmvbkbc, 2023-09-12:
> What is the minimum for writing my own driver?

Minimum what? Minimum understanding: understanding the gpio driver structure and understanding the hardware.

zap8600, 2023-09-12:
> Minimum what? Minimum understanding: understanding the gpio driver structure and understanding the hardware.

Minimum my driver needs to provide.

jcmvbkbc, 2023-09-12:
IDK. I don't have any experience with the gpio subsystem whatsoever.

zap8600, 2023-09-27:
Tell me if I'm wrong, but each GPIO pin takes up 2 bytes in the MMIO, so GPIO0 is at `0x60004000`, GPIO1 is at `0x60004004`, etc.

jcmvbkbc, 2023-09-27:
> Is this right?

No. Take a look at the chapter 6.14 "Register Summary" in the [esp32s3 TRM](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf) for the details.

jcmvbkbc, 2023-09-27:
> would this work?

About that pattern: tell me if I'm wrong, but you can compile it, run it and see for yourself?

zap8600, 2023-09-27:
So I did a bit of thinking, and I think I know how to use the GPIO now. I can use GPIO_ENABLE_REG to set the pin I want to output, then pull it high with GPIO_OUT_REG. I'm not sure when I'll be able to test my code, but I'm (sorta) confident in my code.

closed 2023-10-24 by zap8600