jcmvbkbc/linux-xtensa #15: USB Host Controller

zap8600, 2024-01-10:
What would it take for a USB Host Controller driver to be made?

jcmvbkbc, 2024-01-23:
As mentioned https://github.com/jcmvbkbc/linux-xtensa/issues/5#issuecomment-1720040704 there's some figuring out the hardware details from the existing code that must happen. And then the standard: copy a similar driver -- fill in the hardware-specific details.

jcmvbkbc, 2024-07-25:
I've pushed initial version with USB host support [here](https://github.com/jcmvbkbc/linux-xtensa/commits/xtensa-6.10-esp32-usb/). Currently it needs the following change to the esp_hosted that switches the USB PHY connection from the USB serial to the USB controller:
```
diff --git a/esp_hosted_ng/esp/esp_driver/network_adapter/main/linux_boot.c b/esp_hosted_ng/esp/esp_driver/network_adapter/main/linux_boot.c
index 0cf55f3a2389..fc9ae8329b4d 100644
--- a/esp_hosted_ng/esp/esp_driver/network_adapter/main/linux_boot.c
+++ b/esp_hosted_ng/esp/esp_driver/network_adapter/main/linux_boot.c
@@ -125,6 +125,8 @@ static void map_flash_and_go(void)
        extern int g_abort_on_ipc;
        g_abort_on_ipc = 1;
 
+       *(volatile uint32_t *)0x60008120 = 0x180000;
+       *(volatile uint32_t *)0x60039000 = 0x001d5000;
        asm volatile ("mov a2, %1 ; jx %0" :: "r"(linux), "r"(tag) : "a2");
 }
 
```

This version has DMA disabled so the performance is not impressive, my guess is that it would only work with the buffers and/or descriptors in DRAM but some further investigation is needed.

zap8600, 2024-07-25:
Nice! Currently, I'm working on my U-Boot port and a small laptop with the ESP32-S3, so It'll be a minute before I give it a test. The low speeds are kind of expected. With ESP-IDF running alongside Linux, the capabilities are limited. Hopefully I finish up my U-Boot port soon so I can work on removing those limitations.

zap8600, 2024-08-01:
It might be a bit early to ask, but is being a device possible as well? I think Linux has USB OTG support, and I know the ESP32-S3 has it, too. I'm not sure how Linux handles the switch from host to device and vise versa or how the power supply change would be handled.

jcmvbkbc, 2024-08-01:
> is being a device possible as well?

Yes, I tested it as an ethernet gadget, works pretty well as long as packet size stays small. I guess fine-tuning FIFO sizes is needed to make it solid.

zap8600, 2024-08-01:
If both host and peripheral work, then does OTG work? I'd assume that external USB power would be needed. I'm not sure how Linux detects whether it's a host or peripheral, so I'm not sure how to put it together on a board.

jcmvbkbc, 2024-08-02:
> If both host and peripheral work, then does OTG work?

My impression is that it wouldn't work with the internal USB PHY, but it might work with the external USB PHY.
On the [devkit-c1 board](https://dl.espressif.com/dl/schematics/SCH_ESP32-S3-DevKitC-1_V1.1_20220413.pdf) it 100% doesn't work through the onboard USB connector because the ID line is not connected to it. Even the host mode doesn't work for the bus-powered devices through the onboard USB connector.

zap8600, 2024-08-02:
I thought that the ESP32-S3 had GPIO pins for detecting A-mode or B-mode. I think they're called `usb_otg_avalid_in` and `usb_otg_bvalid_in` in the TRM. Could those be used, or am I wrong?

jcmvbkbc, 2024-08-03:
> Could those be used

They can be used, but you need something that would drive them. Internal USB PHY cannot do it.
In my configurations I [statically set them high or low](https://github.com/jcmvbkbc/linux-xtensa/commit/ee8214d2cd12d68c721c18898747e59334228e06#diff-bcd29b0189412c4b8ce39f1e2c70a794c1268d58b58c0d0232f891045e021cc5R29-R43) depending on whether the host or the device mode is configured in the DTS.

zap8600, 2024-08-03:
I thought they were inputs, but I guess I was wrong. Sorry. In that case, then how does the external transceiver work? I'm not sure I understand it.

jcmvbkbc, 2024-08-03:
> I thought they were inputs

They are.