jcmvbkbc/linux-xtensa #19: Booting Linux on ESP32-S3 with U-Boot

zap8600, 2024-04-13:
Would it be possible to get a proper bootloader like U-Boot onto the ESP32-S3 and boot Linux with it?

jcmvbkbc, 2024-04-13:
It would, but why would anybody want that?

zap8600, 2024-04-13:
I'm interested in trying to create drivers for the ESP32-S3's hardware without the need of ESP-IDF or without anything that ESP-IDF does getting in the way. I do also want to check out developing for bare-metal Xtensa on the ESP32-S3.

max1220, 2024-04-17:
Developing your own drivers will be hard for the Wifi part, which requires closed-source firmware that interacts with FreeRTOS(or at least tries to; You could possibly stub out the required FreeRTOS functions). The Wifi hardware is not documented. I think in some ways the ESP-IDF makes creating new drivers for Linux easier - You can just create a small binding between FreeRTOS and Linux via the IPC mechanism, instead of a full driver. If you actually just want working drivers, don't re-implement them yourself, just use the ESP-IDF drivers in Linux via the IPC, and write a Linux "stub" driver. But if you just want to learn about embedded programming I guess you could just try to port u-boot! Learning is guaranteed :P

zap8600, 2024-04-28:
I will learn, which is what I'm aiming for! I want to see if I can get U-Boot to replace the main app entirely so that only the bare minimum of ESP-IDF would be needed.

zap8600, 2024-12-10:
Sorry, can't say I've made much progress. I've been caught up with school and work, and before that, I was struggling to get the fundamentals of the processor to work properly. Plus, I still need to make a custom bare-bones bootloader that sets up the bare minimum for operating systems. Sorry to disappoint. As for kernel sources, this repository on the `xtensa-6.11-esp32` branch should work.

For peripherals, you have GPIO, USB, and SPI. You can bit-bang I2C.

erosenst, 2024-12-10:
Thanks for fast reply, I just cloned U-boot sources and have fresh 19.1.5 clang (Xtensa&etc) compiled to /usr/local 
 I'll clone 6.11-esp32 it is for sx6 cores.? I have sx7 chips but they have not board soldered to them so little difficult to use..

Wifi drivers need to get from esp-idf for closed source blob reason.? Is there any reverse engineering or some docs of them.? Got some google link that someone have done hacks of them but no documentation how..

Thank you

zap8600, 2025-02-08:
Closing this. I'm going to work on adding drivers to Linux and better understanding the hardware before I finish this.

closed 2025-02-08 by zap8600